dartsim/dart #1536: Mac OSX pip install dartpy ipopt issue

jaredmoore, 2021-02-26:
> If you're not reporting a bug, please use the [forum](https://dartsim.discourse.group/) to ask questions.

### Bug Report
- [X] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [X] I checked to make sure that this issue has not already been filed.

#### Environment
* DART version: master
* OS name and version name(or number): macOS Catalina
* Compiler name and version number: Apple clang version 12.0.0 (clang-1200.0.32.21)

#### Expected Behavior
Trying to follow the Mac OS installation instructions using pip3 install dartpy inside a virtual environment configured with Python 3.9.1.

#### Current Behavior
Command pip3 install dartpy fails at 48% on the ipopt installation again.  It appears to be another occurrence of the issue in https://github.com/dartsim/dart/issues/1504

[gist output](https://gist.github.com/jaredmoore/ae013d90fa801ea84050db11c23ea6dc)

#### Steps to Reproduce

1. brew install dartsim --only-dependencies
2. brew install pybind11
3. pip3 install dartpy

Could it potentially be that the pip wheel hasn't been updated since the recent change?

mhortho44, 2023-09-28:
Any resolution to this?

jslee02, 2023-12-13:
`pip install dartpy` has been updated and this issue is no longer reproducible. Closing for now, but feel free to reopen if the issue persists.

closed 2023-12-13 by jslee02