dartsim/dart #1554: eigen: too many iterations in Jacobi transform.

trns1997, 2021-04-11:
Hey guys,
I was playing `./tutorial_collisions_finished` tutorial to test soft bodies. I successfully was able to create soft bodies for box shape, cylinder shape and spherical shape.

When I create the soft box and soft cylinder they work as expected:

[image]

But When I test with the sphere i get this error and the ball seems to explode or disappear :
```
eigen: too many iterations in Jacobi transform.
```
[image]

Here is the code for each soft body object:
```
SkeletonPtr createSoftBox()
{
  SkeletonPtr soft = Skeleton::create("soft");

  // Add a soft body
  BodyNode* bn = addSoftBody<FreeJoint>(soft, "soft box", SOFT_BOX);

  // Add a rigid collision geometry and inertia
  double width = default_shape_height, height = 2 * default_shape_width;
  Eigen::Vector3d dims(width, width, height);
  dims *= 0.6;
  std::shared_ptr<BoxShape> box = std::make_shared<BoxShape>(dims);
  bn->createShapeNodeWith<VisualAspect, CollisionAspect, DynamicsAspect>(box);

  Inertia inertia;
  inertia.setMass(default_shape_density * box->getVolume());
  inertia.setMoment(box->computeInertia(inertia.getMass()));
  bn->setInertia(inertia);

  setAllColors(soft, dart::Color::Fuchsia());

  return soft;
}

SkeletonPtr createSoftBall()
{
  SkeletonPtr soft = Skeleton::create("softball");

  // Add a soft body
  BodyNode* bn = addSoftBody<FreeJoint>(soft, "soft sphere", SOFT_ELLIPSOID);

  double radius = default_shape_height / 2.0;
  Eigen::Vector3d dims = 2 * radius * Eigen::Vector3d::Ones();

  dims *= 0.6;
  std::shared_ptr<EllipsoidShape> sphere = std::make_shared<EllipsoidShape>(dims);
  bn->createShapeNodeWith<VisualAspect, CollisionAspect, DynamicsAspect>(sphere);

  Inertia inertia;
  inertia.setMass(default_shape_density * sphere->getVolume());
  inertia.setMoment(sphere->computeInertia(inertia.getMass()));
  bn->setInertia(inertia);

  setAllColors(soft, dart::Color::Fuchsia());

  return soft;
}

SkeletonPtr createSoftCylinder()
{
  SkeletonPtr soft = Skeleton::create("softcylinder");

// Add a soft body
  BodyNode* bn = addSoftBody<FreeJoint>(soft, "soft cylinder", SOFT_CYLINDER);

  double radius = default_shape_height / 2.0,
           height = 2 * default_shape_width;

  radius *= 0.6;
  height *= 0.6;
  std::shared_ptr<CylinderShape> cylinder = std::make_shared<CylinderShape>(radius, height);
  bn->createShapeNodeWith<VisualAspect, CollisionAspect, DynamicsAspect>(cylinder);

  Inertia inertia;
  inertia.setMass(default_shape_density * cylinder->getVolume());
  inertia.setMoment(cylinder->computeInertia(inertia.getMass()));
  bn->setInertia(inertia);

  setAllColors(soft, dart::Color::Fuchsia());

  return soft;
}
```

If you need anything else just let me know. Hopefully someone could help me understand what causes this error.

jslee02, 2021-04-14:
Which DART version did you use? I was able to throw soft sphere bodies with `master` branch: https://github.com/dartsim/dart/tree/test_soft_ellipsoids

[image]

trns1997, 2021-04-14:
@jslee02 thanks for the branch it seems to work perfectly fine, I tried again on v6.9.5 and now it works maybe i had a sneaky little typo somewhere earlier that caused the error. Also I'd like to know what causes this error, because I am trying to get the softbody nodes working with gazebo 11 and i tend to get the same error for sphere shape. Thanks btw.

jslee02, 2021-04-14:
Glad it works now! I'm closing this issue, but feel free to reopen if there is any finding that you'd like to share.

closed 2021-04-14 by jslee02

mxgrey, 2021-04-14:
> I'd like to know what causes this error

I think the error likely happens when the soft body gets deformed into a state that causes some kind of singularity in the constraint solver. If version changes are affecting things, I might suspect something about the collision detection is different. Maybe the collision detection for the failed version is doing a bad job of reporting collisions (e.g. not reporting enough collisions, or not calculating the normal vector in a good way) and that allows the body to deform to a point where the constraint solver can't find a good solution for the physics constraints.