dartsim/dart #1557: Default Front Face Culling?

lakshmipathyarjun6, 2021-04-13:
Firstly great simulator - really appreciate the work!

Out of curiosity is there a reason that front face culling is enabled by default in https://github.com/dartsim/dart/blob/08a75807049669b26f5c175eb936e843d34bd945/dart/gui/OpenGLRenderInterface.cpp#L76? I kept suspecting there was a bug in my mesh normals but was admittedly a bit surprised / was expecting back face culling by default. 

Not sure if this was intentional so didn't want to unnecessarily report a bug.

jslee02, 2021-04-14:
This is definitely a bug. It seems this issue hasn't been catched because most of the meshes we've used has no `AI_MATKEY_TWOSIDED` so it's always disabled:

https://github.com/dartsim/dart/blob/08a75807049669b26f5c175eb936e843d34bd945/dart/gui/OpenGLRenderInterface.cpp#L616-L621

This should be fixed by https://github.com/dartsim/dart/pull/1559

lakshmipathyarjun6, 2021-04-14:
Thanks for the fast reply and quick fix!

It was definitely an atypical case - I basically had a mesh with a specific vertex coloring that I needed to preserve but did not construct a material for, so I removed the default material and bypassed that check entirely

jslee02, 2021-04-14:
Yeah, it was a good catch! Please let me know if the issue is still persistent even with #1559.

PR #1559 (2021-04-14): Set back face culling by default
Fixes #1557
closed 2021-04-14 by jslee02