dartsim/dart #1603: setPosition() function bug

ZZZZZAF, 2021-08-30:
Hi, I'm using dartpy.

In biped_stand example, when I run this two statements below, the biped didn't rotate as I expected(I think it should rotate 90 degree w.r.t x axis, then rotate -90 degree w.r.t y axis, bug it didn't behavior as I thought)
    biped.getDof('j_pelvis_rot_x').setPosition(1.5708)
    biped.getDof('j_pelvis_rot_y').setPosition(-1.5708)

Is this a bug?
Please help me!

mxgrey, 2021-08-30:
The 3 rotational degrees of freedom of a `FreeJoint` (which has 6 degrees of freedom in total, the other 3 being translational) use a logmap representation for the rotation. They do not use Euler angles since Euler angles are vulnerable to gimbal lock.

The easiest way to do what you want is to use [`dart::dynamics::FreeJoint::convertToPositions(tf)`](https://github.com/dartsim/dart/blob/552f8d155f7d9c9ea6dd1d02cf63631ac290dfe3/dart/dynamics/FreeJoint.hpp#L80-L86) and then pass those 6 position values into the `dart::dynamics::Joint::setPositions(~)` function of your root `FreeJoint`.

I'll be closing this ticket as this is the intended behavior, not a bug.

closed 2021-08-30 by mxgrey

ZZZZZAF, 2021-08-30:
Thank you very much