dartsim/dart #1604: Default Inertia mass equal to 1kg

costashatz, 2021-09-07:
This is not exactly a bug (nor a feature request).

### Bug Report

- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment

* DART version: master (6.11, 6.10, and I guess most others)
* OS name and version name(or number): does not matter
* Compiler name and version number: does not matter

#### Expected Behavior
When reading URDF, there are a few links that do not define inertia properties. The current default behavior of DART is to assume a mass of 1kg and identity moment of inertia (see [here](https://github.com/dartsim/dart/blob/main/dart/dynamics/Inertia.hpp#L68-L71)). The issue with this is that in big URDFs that we do not make (e.g., Talos humanoid) there can be several hidden links like this and then the simulation behaves weirdly (this is because usually these links are "dummy" links to create some complex structure or for creating frames). Right now I have to check all URDFs and clean them from all these links.

I suggest giving either `Inertia` a different default initialization or changing the URDF parser to include a "dummy" inertia when there is no inertia tag: what I usually do is put put small mass (1e-8) and the moments of inertia to identity*mass. This gives normal behavior.

#### Steps to Reproduce
1.  Take any URDF with a link with no inertia tag
2.  Load it in DART
3.  Observe the simulation
4.  Or print the mass/moment of inertia

jslee02, 2021-09-18:
How about we let the user decides which values to use for the missing properties? Changing the default values could still be a problem for other usesr (e.g., someone could consider 1e-8 is still too large or too small value). #1605 is a tentative solution to this.

costashatz, 2021-09-18:
This is very good. Thanks a lot. If you can merge this also in version 6.11.1+ I'd appreciate it (I haven't still restructured my code for DART master v7+).

jslee02, 2021-09-18:
I'm thinking of merging this to 6.12.0, which is the current `main` branch. DART v7 is in `dart7/*` branches. Would this work for you?

costashatz, 2021-09-18:
I guess so.. I remember the `main` to be for DART 7, but I guess you changed this. So 6.12.0 (assuming that the API is the same as 6.11.x) would be fine for me.

jslee02, 2021-09-18:
Great. Yeah, it's changed because it would take some time for DART 7 to be ready for release. I thought it'd be good to keep DART 6 as the mainstream (so in `main`) until then. Sorry for any confusion!

costashatz, 2021-09-18:
No worries. If you can update the issue when the PR is merged. Thanks a lot!

PR #1605 (2021-09-19): Add default options to DartLoader for missing properties in URDF files
Enable to pass parsing options to `DartLoader`. Currently the options are default resource retriever, default root joint type, and default inertia of link. The default inertia options is to resolve #1604.

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [x] Add unit test(s) for this change
closed 2021-09-19 by jslee02