dartsim/dart #1618: Does dartpy support closed loop(parallel) structure/mechanism?

lisheng-k, 2021-11-01:
Hi, 

I have to raise an issue here. it's about closed-loop (parallel) structure in dartpy.

I descrived a parallel structure (a delta robot) in sdf file, then tried to parser the file with dartpy.

but the error came out and said :
"
Error [SdfParser.cpp:1138] [SdfParser::readAllJoints] Joint named [rear2_platform] is claiming Link [platform] as its child, but that is already claimed by Joint [rear1_platform]. Joint [rear2_platform] will be discarded
Error [SdfParser.cpp:1138] [SdfParser::readAllJoints] Joint named [rear3_platform] is claiming Link [platform] as its child, but that is already claimed by Joint [rear1_platform]. Joint [rear3_platform] will be discarded
"

looks like a link can only be one 'child' to one 'parent' link.

Is there a specific way to connect a child link to more than one parent link?

or, does dartpy support this kind of structure.

The structure I described in the sdf is -- a mobile platform linked to three separate links with three ball joints.

image see bellow:

[image]

thanks in advance !

> If you're not reporting a bug, please use the [forum](https://dartsim.discourse.group/) to ask questions.

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [ ] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [ ] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: [e.g., master, 6.8.3]
* OS name and version name(or number): [e.g., Ubuntu 18.04, macOS Catalina, Windows 10]
* Compiler name and version number: [e.g., GCC 7.4.0, Clang 3.9.1]

#### Expected Behavior
> Please describe the behavior you are expecting.

#### Current Behavior
> What is the current behavior?
> If you have trouble in `cmake` or `make`, it would be very helpful to build DART with verbose option:
> `$ cmake .. -DDART_VERBOSE=On`
> `$ make VERBOSE=1`
> and then dump the results into a [gist](https://gist.github.com/) and share the link to gist here.

#### Steps to Reproduce
> Please provide detailed steps for reproducing the issue.
1. 
2. 
3. 
4. 

#### Code to Reproduce
> Please remember that with a working code it's easier to reproduce the bug and it's much faster to fix it.
> It would be great to fork this repo and create a regression test using [this template](https://github.com/dartsim/dart/blob/master/unittests/regression/test_Issue000Template.cpp).

jslee02, 2021-11-01:
DART supports closed loop structure, but the current SDF parser doesn't. You could form a closed loop structure by adding dynamic joint constrains in code. Here is an example: https://github.com/dartsim/dart/blob/b663c87e73e3c48295a856f22d5f455be9dc3443/python/examples/rigid_loop/main.py

lisheng-k, 2021-11-01:
> DART supports closed loop structure, but the current SDF parser doesn't. You could form a closed loop structure by adding dynamic joint constrains in code. Here is an example: https://github.com/dartsim/dart/blob/b663c87e73e3c48295a856f22d5f455be9dc3443/python/examples/rigid_loop/main.py

great thanks, 
I will try that !

closed 2021-12-24 by jslee02