dartsim/dart #1654: Capsule shapes sink through the ground with ODE collision detector

azeey, 2022-01-25:
### Bug Report
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
* DART version: `main`
* OS name and version name(or number): Ubuntu 20.04
* Compiler name and version number: GCC 9.3

#### Expected Behavior
I expect the capsule shape to stay above the ground plane/box in all orientations of the capsule.

#### Current Behavior

A capsule on it's side placed on the ground slowly sinks through the ground. This doesn't happen if the capsule is vertical and only one hemisphere is in contact with the ground.

A similar problem was fixed in https://github.com/dartsim/dart/pull/859 for Bullet as the collision detector. Would it be possible to apply something similar to ODE?

Related:
https://github.com/ignitionrobotics/ign-gazebo/issues/1306

#### Steps to Reproduce
1. Build code and tests from fork (branch `azeey/issue1654`)
2. Run `test_Issue1654`.

#### Code to Reproduce

https://github.com/ignition-forks/dart/tree/azeey/issue1654

azeey, 2024-12-16:
@jslee02 any chance you can take a look at this or see if #1832 is a viable solution?

jslee02, 2024-12-30:
I'm afraid to say, but currently I don't have enough bandwidth to thoroughly investigate the issue and review the details of the PR for it at the moment. However, I'd be happy to accept the PR if it demonstrates a solution with sufficient testing, showing that it resolves the issue without introducing new problems, and doesn't significantly impact existing features or performance.