dartsim/dart #1655: 6.12.1 fails to find pagmo and flann

yurivict, 2022-01-31:
pagmo2-2.18.0 and flann-1.9.1.29 are installed but it fails to find them:
```
CMake Warning at cmake/DARTMacros.cmake:145 (message):
  Looking for pagmo - NOT found, to use dart-optimizer-pagmo, please install
  pagmo
Call Stack (most recent call first):
  dart/optimizer/pagmo/CMakeLists.txt:3 (dart_check_optional_package)

-- Performing Test BT_USE_DOUBLE_PRECISION
-- Performing Test BT_USE_DOUBLE_PRECISION - Failed
CMake Warning at cmake/DARTMacros.cmake:145 (message):
  Looking for flann - NOT found, to use dart-planning, please install flann
Call Stack (most recent call first):
  dart/planning/CMakeLists.txt:3 (dart_check_optional_package)

```

olzhas, 2022-07-26:
what OS are you using and how did you install those packages?

yurivict, 2022-07-26:
FreeBSD.

All packages are installed from packages.

olzhas, 2022-07-28:
can't help with this, I don't have FreeBSD at my hand.

jslee02, 2023-01-07:
No plan to fix this for DART 6.12.1, but finding pagmo2 is fixed in `master` (DART 7). FLANN is no longer a dependency for DART 7.

closed 2023-01-07 by jslee02