dartsim/dart #1671: [urdfdom] Build failure on Windows with urdfdom version 3.1.0

Thomas1664, 2022-07-06:
> If you're not reporting a bug, please use the [forum](https://dartsim.discourse.group/) to ask questions.

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: 6.12.1
* OS name and version name(or number): Windows 10
* Compiler name and version number: MSVC with Visual Studio 2022

I tried to update the `urdfdom` port from vcpkg to 3.1.0 in this PR: https://github.com/microsoft/vcpkg/pull/25591

On Windows, build fails with the following error message:

```
FAILED: dart/utils/urdf/CMakeFiles/dart-utils-urdf.dir/DartLoader.cpp.obj 
C:\PROGRA~1\MICROS~1\2022\ENTERP~1\VC\Tools\MSVC\1432~1.313\bin\Hostx64\x64\cl.exe   /TP -DBOOST_ALL_NO_LIB -DBOOST_TEST_DYN_LINK -DTINYXML2_DEBUG -DTINYXML2_IMPORT -D_CRT_SECURE_NO_WARNINGS -D_ENABLE_EXTENDED_ALIGNED_STORAGE -ID:\buildtrees\dartsim\src\v6.12.1-cbdec63edf.clean -ID:\buildtrees\dartsim\x64-windows-dbg -external:ID:\installed\x64-windows\include\eigen3 -external:ID:\installed\x64-windows\include -external:ID:\installed\x64-windows\include\utf8cpp -external:ID:\installed\x64-windows\include\irrlicht -external:ID:\installed\x64-windows\include\kubazip -external:ID:\installed\x64-windows\include\poly2tri -external:ID:\installed\x64-windows\share\urdfdom_headers\..\..\include -external:ID:\installed\x64-windows\include\urdfdom -external:ID:\installed\x64-windows\include\urdfdom_headers -external:W0 /nologo /DWIN32 /D_WINDOWS /W3 /utf-8 /GR /EHsc /MP  /EHsc /permissive- /Zc:twoPhase- /D_DEBUG /MDd /Z7 /Ob0 /Od /RTC1  /wd4005 /wd4099 /wd4244 /wd4250 /wd4267 /wd4305 /wd4334 /wd4838 /wd4996 /bigobj /showIncludes /Fodart\utils\urdf\CMakeFiles\dart-utils-urdf.dir\DartLoader.cpp.obj /Fddart\utils\urdf\CMakeFiles\dart-utils-urdf.dir\dart-utils-urdf.pdb /FS -c D:\buildtrees\dartsim\src\v6.12.1-cbdec63edf.clean\dart\utils\urdf\DartLoader.cpp
D:\installed\x64-windows\include\urdfdom_headers\urdf_sensor/sensor.h(93): error C2208: 'double': no members defined using this type
D:\installed\x64-windows\include\urdfdom_headers\urdf_sensor/sensor.h(94): error C2208: 'double': no members defined using this type
D:\installed\x64-windows\include\urdfdom_headers\urdf_sensor/sensor.h(102): error C2059: syntax error: '='
D:\installed\x64-windows\include\urdfdom_headers\urdf_sensor/sensor.h(103): error C2059: syntax error: '='
```

I don't have any idea what is causing the build failure since declaring a `double` member shouldn't be an issue here. And I don't even see the include path for the header.

jslee02, 2022-07-31:
I believe this is because `near` and `far` are reserved by `windows.h`, and it's fixed by https://github.com/dartsim/dart/pull/1666, which is not released yet. Let me backport the fix for v6.12.2.

Thomas1664, 2022-07-31:
> I believe this is because `near` and `far` are reserved by `windows.h`, and it's fixed by #1666, which is not released yet. Let me backport the fix for v6.12.2. In the meantime, feel free to create a patch to vcpkg out of #1666.

@jslee02 I just used the latest commit because the patch of #1666 did not apply. But now I get a different error on x86 Windows:

```
FAILED: dart/CMakeFiles/dart.dir/common/PoolAllocator.cpp.obj 
C:\PROGRA~1\MICROS~1\2022\ENTERP~1\VC\Tools\MSVC\1432~1.313\bin\Hostx64\x86\cl.exe   /TP -DDART_ACTIVE_LOG_LEVEL=1 -DDART_HAVE_spdlog=0 -DFMT_HEADER_ONLY=1 -D_CRT_SECURE_NO_WARNINGS -D_ENABLE_EXTENDED_ALIGNED_STORAGE -ID:\buildtrees\dartsim\src\ff091f46b3-95699f332c.clean -ID:\buildtrees\dartsim\x86-windows-dbg -external:ID:\installed\x86-windows\include\eigen3 -external:ID:\installed\x86-windows\include -external:W0 /nologo /DWIN32 /D_WINDOWS /W3 /utf-8 /GR /EHsc /MP  /EHsc /permissive- /Zc:twoPhase- /D_DEBUG /MDd /Z7 /Ob0 /Od /RTC1  /wd4005 /wd4099 /wd4146 /wd4244 /wd4250 /wd4267 /wd4305 /wd4334 /wd4838 /wd4996 /bigobj -std:c++17 /showIncludes /Fodart\CMakeFiles\dart.dir\common\PoolAllocator.cpp.obj /Fddart\CMakeFiles\dart.dir\dart.pdb /FS -c D:\buildtrees\dartsim\src\ff091f46b3-95699f332c.clean\dart\common\PoolAllocator.cpp
D:\buildtrees\dartsim\src\ff091f46b3-95699f332c.clean\dart\common\PoolAllocator.cpp(48): error C2338: static_assert failed: 'sizeof(MemoryUnit) should be equal to or greater than 8.'
```

Is this intended and compiling in 32 bit mode no longer supported or is this a bug?

jslee02, 2022-07-31:
> Is this intended and compiling in 32 bit mode no longer supported or is this a bug?

It's intended. DART used to support and be tested in 32-bit up to 6.8.5 (only on GCC), but not afterward. We could reconsider this if needed, but 32-bit is not currently supported. Would it be possible to disable the 32-bit build for DART in vcpkg?

Thomas1664, 2022-07-31:
>  Would it be possible to disable the 32-bit build for DART in vcpkg?

Yes. that's possible. Thanks for your help!

jslee02, 2022-07-31:
FYI, just created a new tag ([v6.12.2](https://github.com/dartsim/dart/releases/tag/v6.12.2)), so feel free to use it for the vcpkg receipt. Thank you for updating the dartsim receipt!

PR #1675 (2022-07-31): Fix windows near far
Fixes #1671 

***

#### Before creating a pull request

- [ ] Document new methods and classes
- [ ] Format new code files using ClangFormat by running `make format`
- [ ] Build with `-DDART_TREAT_WARNINGS_AS_ERRORS=ON` and resolve all the compile warnings

#### Before merging a pull request

- [ ] Set version target by selecting a milestone on the right side
- [ ] Summarize this change in `CHANGELOG.md`
- [ ] Add unit test(s) for this change
- [ ] Add Python bindings for new methods and classes
closed 2022-07-31 by jslee02