dartsim/dart #1684: Feature request : Supply axis index when specifying mimic joint.

adityapande-1995, 2022-11-17:
**Is your feature request related to a problem? Please describe.**
When specifying mimic joint (https://dartsim.github.io/dart/main/d6/d5b/classdart_1_1dynamics_1_1Joint.html#ae65b6209299935e602b675f84091c499)

```
 void dart::dynamics::Joint::setMimicJoint( 
        const Joint* _mimicJoint, double _mimicMultiplier, double _mimicOffset)
```

Only the joint pointer, multiplier, and offset are supplied to the method. The section here : (https://github.com/dartsim/dart/blob/189e24d5624f3007fe13d2c27bbdb9a83ed135ec/dart/constraint/MimicMotorConstraint.cpp#L67) dictates that the degrees of freedom for both axes should be the same.

This does not allow mimicking individual axes of a multi-axis joint, such as enabling a revolute joint to mimic one axis of an universal joint, or have an universal joint mimic only one axis of another universal joint. In such cases, it should be permitted to specify the axis index to be mimicked for each joint, and expand the implementation of the mimic joint to accommodate this situation. 

**Describe the solution you'd like**
For instance, the new method could look like : 
```
 void dart::dynamics::Joint::setMimicJoint( const Joint* _mimicJoint, std::size_t _index  , double _mimicMultiplier, double _mimicOffset)
```

**Additional context**
Originally came up here : https://github.com/gazebosim/gz-physics/pull/431

cc @scpeters

scpeters, 2023-05-05:
I'll expand on Aditya's API proposals. (Note that in our proposal for adding a mimic constraint to SDFormat, we have started using "follower" and "leader" terminology, so I will use that same terminology here.)

At a minimum, we'd like to mimic an individual Dof of a multi-Dof joint. So that API could look like:

~~~
void dart::dynamics::Joint::setMimicJoint(
  std::size_t _followerDof,
  const Joint* _leaderJoint,
  std::size_t _leaderDof,
  double _mimicMultiplier,
  double _mimicOffset);
~~~

But that only supports mimicry of a single Dof and the current MimicConstraint supports mimicry of multiple Dofs, so we could extend the `_followerDof` and `_leaderDof` parameters from scalars to a map:

~~~
void dart::dynamics::Joint::setMimicJoint(
  std::map<std::size_t, std::size_t> _followerToLeaderDofsMap,
  const Joint* _leaderJoint,
  double _mimicMultiplier,
  double _mimicOffset);
~~~

jslee02, 2023-05-13:
Thank you for the detailed description! The feature request appears well-defined and feasible. I've attempted to address this in PR #1752.

scpeters, 2023-06-13:
I left a comment at https://github.com/dartsim/dart/pull/1752#issuecomment-1588462955 about a potential need for further API changes in `Joint::setActuatorType`