dartsim/dart #1741: Getting Contact Force

Cfather, 2023-02-13:
Hi,

I have a quick question. Is it possible to get the contact force (at one contact point) between two bodies (let's say, the robot and the ground) in DART? I checked ContactConstraint.cpp but was still not sure which part of the code I should look at.

Thanks.

jslee02, 2023-02-14:
In DART, contact forces are not directly stored in `ContactConstraint` nor `BodyNode`, but rather in `CollisionResult`, which is produced by the `ConstraintSolver`. Therefore, to obtain contact forces, one needs to first call `getLastCollisionResult()` on `ConstraintSolver`, and then iterate through the contacts in the `CollisionResult` object to extract the relevant forces. Here is an example code that shows how to extract contact forces acting on a `robot_body` in contact with the `ground_body`:

```cpp
auto cs = world->getConstraintSolver();
auto& result = cs->getLastCollisionResult();
if (!result.inCollision(robot_body) || result.inCollision(ground_body)) {
  return;
}
for (auto i = 0; i < result.getNumContacts(); ++i) {
  const auto& contact = result.getContact(i);
  if (auto sn1 = contact.collisionObject1->asShapeNode() && sn1->getBodyNodePtr() == robot_body) {
    if (auto sn2 = contact.collisionObject2->asShapeNode() && sn2->getBodyNodePtr() == ground_body) {
      contact.force;  // contact force acting on robot_body in the world coordinate
    }
  }
}
```

Cfather, 2023-02-14:
Thanks for your reply!

closed 2023-02-18 by jslee02

omerkayan, 2023-09-03:
I get 2 different errors. Could you please help me?
error: ‘class dart::collision::CollisionObject’ has no member named ‘asShapeNode’
error: use of ‘sn2’ before deduction of ‘auto’