dartsim/dart #1744: geate install

mbark125, 2023-03-08:
In file included from /snap/root-framework/919/usr/local/include/ROOT/RConfig.hxx:23,
                 from /snap/root-framework/919/usr/local/include/RtypesCore.h:23,
                 from /snap/root-framework/919/usr/local/include/Rtypes.h:23,
                 from /snap/root-framework/919/usr/local/include/TObject.h:17,
                 from /snap/root-framework/919/usr/local/include/TNamed.h:25,
                 from /snap/root-framework/919/usr/local/include/TDirectory.h:24,
                 from /snap/root-framework/919/usr/local/include/TROOT.h:28,
                 from /home/fadil/pet/Gate/Gate-9.0/source/arf/include/GateARFDataToRoot.hh:15,
                 from /home/fadil/pet/Gate/Gate-9.0/source/arf/src/GateARFDataToRoot.cc:13:
/snap/root-framework/919/usr/local/include/RConfigure.h:30:4: warning: #warning "The C++ standard in this build does not match ROOT configuration (201703L); this might cause unexpected issues" [-Wcpp]
   30 | #  warning "The C++ standard in this build does not match ROOT configuration (201703L); this might cause unexpected issues"
      |    ^~~~~~~
In file included from /snap/root-framework/919/usr/local/include/ROOT/RConfig.hxx:23,
                 from /snap/root-framework/919/usr/local/include/RtypesCore.h:23,
                 from /snap/root-framework/919/usr/local/include/Rtypes.h:23,
                 from /snap/root-framework/919/usr/local/include/TObject.h:17,
                 from /snap/root-framework/919/usr/local/include/TNamed.h:25,
                 from /snap/root-framework/919/usr/local/include/TDirectory.h:24,
                 from /snap/root-framework/919/usr/local/include/TROOT.h:28,
                 from /home/fadil/pet/Gate/Gate-9.0/source/arf/include/GateARFDataToRoot.hh:15,
                 from /home/fadil/pet/Gate/Gate-9.0/source/arf/include/GateARFDataToRootMessenger.hh:17,
                 from /home/fadil/pet/Gate/Gate-9.0/source/arf/src/GateARFDataToRootMessenger.cc:12:
/snap/root-framework/919/usr/local/include/RConfigure.h:30:4: warning: #warning "The C++ standard in this build does not match ROOT configuration (201703L); this might cause unexpected issues" [-Wcpp]
   30 | #  warning "The C++ standard in this build does not match ROOT configuration (201703L); this might cause unexpected issues"
      |    ^~~~~~~
In file included from /snap/root-framework/919/usr/local/include/ROOT/RConfig.hxx:23,
                 from /snap/root-framework/919/usr/local/include/RtypesCore.h:23,
                 from /snap/root-framework/919/usr/local/include/Rtypes.h:23,
                 from /snap/root-framework/919/usr/local/include/TObject.h:17,
                 from /snap/root-framework/919/usr/local/include/TNamed.h:25,
                 from /snap/root-framework/919/usr/local/include/TDirectory.h:24,
                 from /snap/root-framework/919/usr/local/include/TROOT.h:28,
                 from /home/fadil/pet/Gate/Gate-9.0/source/arf/include/GateARFSD.hh:19,
                 from /home/fadil/pet/Gate/Gate-9.0/source/arf/src/GateARFSDMessenger.cc:11:
/snap/root-framework/919/usr/local/include/RConfigure.h:30:4: warning: #warning "The C++ standard in this build does not match ROOT configuration (201703L); this might cause unexpected issues" [-Wcpp]
   30 | #  warning "The C++ standard in this build does not match ROOT configuration (201703L); this might cause unexpected issues"
      |    ^~~~~~~
In file included from /snap/root-framework/919/usr/local/include/ROOT/RConfig.hxx:23,
                 from /snap/root-framework/919/usr/local/include/RtypesCore.h:23,
                 from /snap/root-framework/919/usr/local/include/Rtypes.h:23,
                 from /snap/root-framework/919/usr/local/include/TObject.h:17,
                 from /snap/root-framework/919/usr/local/include/TNamed.h:25,
                 from /snap/root-framework/919/usr/local/include/TDirectory.h:24,
                 from /snap/root-framework/919/usr/local/include/TROOT.h:28,
                 from /home/fadil/pet/Gate/Gate-9.0/source/arf/include/GateARFSD.hh:19,
                 from /home/fadil/pet/Gate/Gate-9.0/source/arf/src/GateARFSD.cc:12:
/snap/root-framework/919/usr/local/include/RConfigure.h:30:4: warning: #warning "The C++ standard in this build does not match ROOT configuration (201703L); this might cause unexpected issues" [-Wcpp]
   30 | #  warning "The C++ standard in this build does not match ROOT configuration (201703L); this might cause unexpected issues"
      |    ^~~~~~~
In file included from /snap/root-framework/919/usr/local/include/TString.h:29,
                 from /snap/root-framework/919/usr/local/include/TNamed.h:26:
/snap/root-framework/919/usr/local/include/ROOT/RStringView.hxx:90:17: error: expected type-specifier
   90 |        operator std::string_view() const { return std::string_view(fData,fLength); }
      |                 ^~~
In file included from /snap/root-framework/919/usr/local/include/TString.h:29,
                 from /snap/root-framework/919/usr/local/include/TNamed.h:26:
/snap/root-framework/919/usr/local/include/ROOT/RStringView.hxx:90:17: error: expected type-specifier
   90 |        operator std::string_view() const { return std::string_view(fData,fLength); }
      |                 ^~~
/snap/root-framework/919/usr/local/include/TString.h:115:13: error: expected type-specifier
  115 |    operator std::string_view() const { return std::string_view(Data(),fExtent); }
      |             ^~~
/snap/root-framework/919/usr/local/include/TString.h:280:32: error: ‘string_view’ in namespace ‘std’ does not name a type
  280 |    explicit TString(const std::string_view &sub);
      |                                ^~~~~~~~~~~
/snap/root-framework/919/usr/local/include/TString.h:280:27: note: ‘std::string_view’ is only available from C++17 onwards
  280 |    explicit TString(const std::string_view &sub);
      |                           ^~~
/snap/root-framework/919/usr/local/include/TString.h:317:37: error: ‘string_view’ in namespace ‘std’ does not name a type
  317 |    TString    &operator=(const std::string_view &s);
[... 197 lines cut ...]
/snap/root-framework/919/usr/local/include/TString.h:845:15: error: ‘Bool_t operator==(const int&, const char*)’ must have an argument of class or enumerated type
  845 | inline Bool_t operator==(const std::string_view &s1, const char *s2)
      |               ^~~~~~~~
/snap/root-framework/919/usr/local/include/TString.h:858:37: error: ‘string_view’ in namespace ‘std’ does not name a type
  858 |   std::string printValue(const std::string_view* val);
      |                                     ^~~~~~~~~~~
/snap/root-framework/919/usr/local/include/TString.h:858:32: note: ‘std::string_view’ is only available from C++17 onwards
  858 |   std::string printValue(const std::string_view* val);
      |                                ^~~
make[2]: *** [CMakeFiles/GateLib.dir/build.make:90 : CMakeFiles/GateLib.dir/source/arf/src/GateARFDataToRootMessenger.cc.o] Erreur 1
make[2]: *** Attente des tâches non terminées....
In file included from /home/fadil/pet/Gate/Gate-9.0/source/arf/include/GateARFSD.hh:20:
/snap/root-framework/919/usr/local/include/TFile.h:329:45: error: ‘std::string_view’ has not been declared
  329 |    static Bool_t       SetCacheFileDir(std::string_view cacheDir, Bool_t operateDisconnected = kTRUE,
      |                                             ^~~~~~~~~~~
/snap/root-framework/919/usr/local/include/TFile.h: In static member function ‘static Bool_t TFile::SetCacheFileDir(ROOT::Internal::TStringView, Bool_t, Bool_t)’:
/snap/root-framework/919/usr/local/include/TFile.h:328:36: error: ‘string_view’ is not a member of ‘std’
  328 |      { return SetCacheFileDir(std::string_view(cacheDir), operateDisconnected, forceCacheread); }
      |                                    ^~~~~~~~~~~
/snap/root-framework/919/usr/local/include/TFile.h:328:36: note: ‘std::string_view’ is only available from C++17 onwards
make[2]: *** [CMakeFiles/GateLib.dir/build.make:76 : CMakeFiles/GateLib.dir/source/arf/src/GateARFDataToRoot.cc.o] Erreur 1
make[2]: *** [CMakeFiles/GateLib.dir/build.make:118 : CMakeFiles/GateLib.dir/source/arf/src/GateARFSDMessenger.cc.o] Erreur 1
make[2]: *** [CMakeFiles/GateLib.dir/build.make:104 : CMakeFiles/GateLib.dir/source/arf/src/GateARFSD.cc.o] Erreur 1
make[1]: *** [CMakeFiles/Makefile2:849 : CMakeFiles/GateLib.dir/all] Erreur 2
make: *** [Makefile:146 : all] Erreur 2

closed 2023-03-08 by jslee02

mbark125, 2023-03-08:
Thank you.
 How can I link it to DART, because it was not installed and I installed it

jslee02, 2023-03-09:
Please check out [dart-examples](https://github.com/dartsim/dart-example), a minimal CMake-based project that links to DART. It should be useful to help you get started. For more advanced examples, please take a look in the [`examples/`](https://github.com/dartsim/dart/tree/v6.13.0/examples) folder.