dartsim/dart #1769: [Issue1445.Collision] Regression test crashes

oysstu, 2023-10-12:
### Bug Report
- [X ] I checked the [documentation](http://dartsim.github.io/) but found no answer.
- [X ] I checked to make sure that this issue has not already been filed.

#### Environment
* DART version: v13.6.0
* OS name and version name(or number): Archlinux
* Compiler name and version number: GNU/GCC 13.2.1

#### Current Behavior
The regression test for #1445 is failing for me on v6.13.0. 
```
[oysstu@os-t14 build]$ ./unittests/regression/test_Issue1445
Running main() from /usr/src/debug/libdart/dart-6.13.0/unittests/gtest/src/gtest_main.cc
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from Issue1445
[ RUN      ] Issue1445.Collision
/usr/include/c++/13.2.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = dart::dynamics::Skeleton::DataCache; _Alloc = Eigen::aligned_allocator<dart::dynamics::Skeleton::DataCache>; reference = dart::dynamics::Skeleton::DataCache&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```

The backtrace when calling from gazebo is as follows, and seem to indicate that it is the same issue as reported in #1445 

```
/usr/include/c++/13.2.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = dart::dynamics::Skeleton::DataCache; _Alloc = Eigen::aligned_allocator<dart::dynamics::Skeleton::DataCache>; reference = dart::dynamics::Skeleton::DataCache&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Stack trace (most recent call last):
#31   Object "/usr/lib/libruby.so.3.0", at 0x7fd17964d24e, in 
#30   Object "/usr/lib/libruby.so.3.0", at 0x7fd17964a43b, in 
#29   Object "/usr/lib/libruby.so.3.0", at 0x7fd179640ec8, in 
#28   Object "/usr/lib/ruby/3.0.0/x86_64-linux/fiddle.so", at 0x7fd178fc7acc, in 
#27   Object "/usr/lib/libruby.so.3.0", at 0x7fd1796131e5, in rb_nogvl
#26   Object "/usr/lib/ruby/3.0.0/x86_64-linux/fiddle.so", at 0x7fd178fc73bc, in 
#25   Object "/usr/lib/libffi.so.8", at 0x7fd175570b72, in ffi_call
#24   Object "/usr/lib/libffi.so.8", at 0x7fd17556df5d, in 
#23   Object "/usr/lib/libffi.so.8", at 0x7fd1755714f5, in 
#22   Object "/usr/lib/libgz-sim8-gz.so.8.0.0", at 0x7fd1754f5c69, in runServer
#21   Object "/usr/lib/libgz-sim8.so.8", at 0x7fd17441be21, in 
#20   Object "/usr/lib/libgz-sim8.so.8", at 0x7fd17442f2ef, in gz::sim::v8::SimulationRunner::Run(unsigned long)
#19   Object "/usr/lib/libgz-sim8.so.8", at 0x7fd17442ebb1, in gz::sim::v8::SimulationRunner::Step(gz::sim::v8::UpdateInfo const&)
#18   Object "/usr/lib/libgz-sim8.so.8", at 0x7fd174426711, in gz::sim::v8::SimulationRunner::UpdateSystems()
#17   Object "/usr/lib/gz-sim-8/plugins/libgz-sim-physics-system.so", at 0x7fd1603318c6, in gz::sim::v8::systems::Physics::Update(gz::sim::v8::UpdateInfo const&, gz::sim::v8::EntityComponentManager&)
#16   Object "/usr/lib/gz-sim-8/plugins/libgz-sim-physics-system.so", at 0x7fd16032012e, in gz::sim::v8::systems::PhysicsPrivate::CreatePhysicsEntities(gz::sim::v8::EntityComponentManager const&, bool)
#15   Object "/usr/lib/gz-sim-8/plugins/libgz-sim-physics-system.so", at 0x7fd16031e9c7, in gz::sim::v8::systems::PhysicsPrivate::CreateModelEntities(gz::sim::v8::EntityComponentManager const&, bool)
#14   Object "/usr/lib/gz-sim-8/plugins/libgz-sim-physics-system.so", at 0x7fd1603a18cc, in void gz::sim::v8::EntityComponentManager::EachNew<gz::sim::v8::components::Component<std::add_lvalue_reference<void>, gz::sim::v8::components::ModelTag, gz::sim::v8::serializers::DefaultSerializer<std::add_lvalue_reference<void> > >, gz::sim::v8::components::Component<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, gz::sim::v8::components::NameTag, gz::sim::v8::serializers::StringSerializer>, gz::sim::v8::components::Component<gz::math::v7::Pose3<double>, gz::sim::v8::components::PoseTag, gz::sim::v8::serializers::DefaultSerializer<gz::math::v7::Pose3<double> > >, gz::sim::v8::components::Component<unsigned long, gz::sim::v8::components::ParentEntityTag, gz::sim::v8::serializers::DefaultSerializer<unsigned long> > >(gz::sim::v8::EntityComponentManager::identity<std::function<bool (unsigned long const&, gz::sim::v8::components::Component<std::add_lvalue_reference<void>, gz::sim::v8::components::ModelTag, gz::sim::v8::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, gz::sim::v8::components::Component<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, gz::sim::v8::components::NameTag, gz::sim::v8::serializers::StringSerializer> const*, gz::sim::v8::components::Component<gz::math::v7::Pose3<double>, gz::sim::v8::components::PoseTag, gz::sim::v8::serializers::DefaultSerializer<gz::math::v7::Pose3<double> > > const*, gz::sim::v8::components::Component<unsigned long, gz::sim::v8::components::ParentEntityTag, gz::sim::v8::serializers::DefaultSerializer<unsigned long> > const*)> >::type) const
#13   Object "/usr/lib/gz-sim-8/plugins/libgz-sim-physics-system.so", at 0x7fd16032a36b, in 
#12   Object "/usr/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7fd14addace4, in virtual thunk to gz::physics::dartsim::SDFFeatures::ConstructSdfNestedModel(gz::physics::Identity const&, sdf::v14::Model const&)
#11   Object "/usr/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7fd14adda4da, in gz::physics::dartsim::SDFFeatures::ConstructSdfModelImpl(unsigned long, sdf::v14::Model const&)
#10   Object "/usr/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7fd14add91dc, in gz::physics::dartsim::SDFFeatures::ConstructSdfJoint(gz::physics::Identity const&, sdf::v14::Joint const&, dart::dynamics::BodyNode*, dart::dynamics::BodyNode*)
#9    Object "/usr/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7fd14adc8d59, in dart::dynamics::RevoluteJoint* dart::dynamics::BodyNode::moveTo<dart::dynamics::RevoluteJoint>(dart::dynamics::BodyNode*, dart::dynamics::RevoluteJoint::Properties const&)
#8    Object "/usr/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7fd14adc8ba0, in dart::dynamics::RevoluteJoint* dart::dynamics::Skeleton::moveBodyNodeTree<dart::dynamics::RevoluteJoint>(dart::dynamics::BodyNode*, std::shared_ptr<dart::dynamics::Skeleton> const&, dart::dynamics::BodyNode*, dart::dynamics::RevoluteJoint::Properties const&)
#7    Object "/usr/lib/libdartd.so.6.13", at 0x7fd14a7686bc, in dart::dynamics::Skeleton::moveBodyNodeTree(dart::dynamics::Joint*, dart::dynamics::BodyNode*, std::shared_ptr<dart::dynamics::Skeleton>, dart::dynamics::BodyNode*)
#6    Object "/usr/lib/libdartd.so.6.13", at 0x7fd14a66d48e, in dart::dynamics::Frame::changeParentFrame(dart::dynamics::Frame*)
#5    Object "/usr/lib/libdartd.so.6.13", at 0x7fd14a64e821, in dart::dynamics::Entity::changeParentFrame(dart::dynamics::Frame*)
#4    Object "/usr/lib/libdartd.so.6.13", at 0x7fd14a61ccc6, in dart::dynamics::BodyNode::dirtyTransform()
#3    Object "/usr/lib/libstdc++.so.6", at 0x7fd1752dd3b1, in std::__glibcxx_assert_fail(char const*, int, char const*, char const*)
#2    Object "/usr/lib/libc.so.6", at 0x7fd1791574b7, in abort
#1    Object "/usr/lib/libc.so.6", at 0x7fd17916f667, in raise
#0    Object "/usr/lib/libc.so.6", at 0x7fd1791bf83c, in 
Aborted (Signal sent by tkill() 3571 1000)
```

#### Steps to Reproduce
The crash can be reproduced in the regression test 1445

Installation was done using the AUR package with the following compilation flags
```
    cmake .. \
        -DCMAKE_BUILD_TYPE='Debug' \
        -DCMAKE_INSTALL_PREFIX="/usr" \
        -DCMAKE_INSTALL_LIBDIR="lib" \
        -DDART_TREAT_WARNINGS_AS_ERRORS="off" \
        -DCMAKE_SKIP_INSTALL_RPATH=ON \
        -DDART_SKIP_FLANN=ON \
        -DDART_SKIP_PAGMO=ON \
        -DDART_VERBOSE=ON \
        -DDART_FAST_DEBUG=OFF
``

And `options=(debug !strip)` to keep debug symbols.

oysstu, 2023-10-12:
There's also a workaround in gz-physics that may or may not be outdated.
https://github.com/gazebosim/gz-physics/blob/1445341181e9b81eaacfee504ccc8fed0f949c9a/dartsim/src/SDFFeatures.cc#L1186-L1190

The original issue was reported in gz-physics here:
https://github.com/gazebosim/gz-physics/pull/268

Let me know if I should raise an issue in gz-physics is this is indeed outdated

oysstu, 2023-10-13:
The crash does not occur on the main/dart7 branch. I tried to go through the commits to see what might be the cause, but found it very hard to pin something down due to the large structural changes for dart7.

oysstu, 2023-10-16:
I've narrowed it down to the following.

Archlinux compiles with `CXXFLAGS="$CFLAGS -Wp,-D_GLIBCXX_ASSERTIONS"` by default. The GLIBCXX_ASSERTIONS flag adds runtime checks for a few additional things, even in release build. One of those is bounds checking for the []-operator. 
In BodyNode::dirtyTransform the mTreeCache vector of Skeleton is accessed using the following macro
https://github.com/dartsim/dart/blob/2d3cbdff352b0236b18c85e06f8047a84b3d1098/dart/dynamics/BodyNode.cpp#L165C1-L169C36

Used from here
https://github.com/dartsim/dart/blob/c5b8f0abfb2545754ebc872e4aad78020da88a62/dart/dynamics/BodyNode.cpp#L1509C1-L1520C4

Since this happens during moveBodyNodeTree, the mTreeCache vector could be in an intermediate state. The vector might have the sufficient room reserved, but not yet resized, or the code might actually access out of bounds memory.

jslee02, 2023-12-13:
Thank you for tracking down this tricky issue! Feel free to submit a PR for this, or I could work on it when I have a chance.

oysstu, 2023-12-14:
Well, I started diving into the problem but the codebase got increasingly more complex, and I had to move on as I'm not familiar with the overall structure of the library. I may revisit this issue in the future if it does not get solved, but I unfortunately don't have time to pursue it right now.

jslee02, 2023-12-14:
Sure, your insights are already helpful, greatly appreciated, and will guide my further investigation. Thank you for your initial efforts!

jslee02, 2023-12-19:
This should be resolved by #1778. Please let me know if this issue persists. Thank you for the error report, again!

PR #1778 (2023-12-19): Attempt to fix #1769
Tested on 7.0 (main) with `-D_GLIBCXX_ASSERTIONS`: https://github.com/dartsim/dart/actions/runs/7258388815/job/19773633871

Resolve #1769 

***

#### Before creating a pull request

- [ ] Document new methods and classes
- [ ] Format new code files using ClangFormat by running `make format`
- [ ] Build with `-DDART_TREAT_WARNINGS_AS_ERRORS=ON` and resolve all the compile warnings

#### Before merging a pull request

- [ ] Set version target by selecting a milestone on the right side
- [ ] Summarize this change in `CHANGELOG.md`
- [ ] Add unit test(s) for this change
- [ ]  [... cut]
commit b7f5dd1c (2023-12-19): Attempt to fix #1769 (#1778)
closed 2023-12-19 by jslee02