dartsim/dart #1771: Collision forces are constantly zero. I expect the total sum to be equal and opposite to the applied force.

Nickick-ICRS, 2023-11-28:
### Bug Report
Dart collision force vector does not match expected applied force. It's constantly zero.

#### Environment
> Select the following information.
* DART version: 6.13 (Ubuntu apt repository)
* OS name and version name(or number): Ubuntu 20.0.4
* Compiler name and version number: GCC 9.4.0

#### Expected Behavior
With a 3 D.O.F simple robot arm (created in the code below), a constant force of 5 N in the Z direction is applied to a fixed plane. This is applied via joint torques using the standard Jacobian transpose method. When contacts are detected, the total sum of contact forces is expected to be equal and opposite (or close to) the applied force.

#### Current Behavior
The contacts in the collision have force equal to zero. Every force individually is zero, and the sum of all forces in the contact is zero.

The force at the end effector is definitely non-zero as the torques are non-zero, and the end effector moves before contact with the plane. As the world gravity vector is zero and there is no motion at contact, there is no gravity or coriolis effect to counteract the received force. Therefore the contact should have a non-zero reaction force.

#### Steps to Reproduce
1. Create a 3 dof robot arm (/leg) with collision aspect for at least the end effector
2. Create a WeldJoint plane (or a FreeJoint plane, with a very large mass and inertia and no gravity)
3. Position the two skeletons such that contact will soon appear when a constant force is applied to the end effector
4. Apply force to the end effector via joint torques (e.g. Jacobian transpose, operational space controller)
5. Detect contacts between the plane and the end effector
6. Sum the forces of the contacts between the plane and end effector and print

#### Code to Reproduce
This needs to be compiled with **osg** to visualise easily:
`#include <dart/dart.hpp>
#include <dart/gui/osg/osg.hpp>

namespace dd = dart::dynamics;
namespace ds = dart::simulation;
namespace dc = dart::collision;

class ApplyForceWorldNode :public dart::gui::osg::WorldNode {
public:
    ApplyForceWorldNode(const ds::WorldPtr &world, bool fixed_plane)
        :dart::gui::osg::WorldNode(world), world_(world)
    {
        const double TIMESTEP = 1e-3;
        world_->setTimeStep(TIMESTEP);
        world_->setGravity(Eigen::Vector3d::Zero());
    
        setup_plane(fixed_plane);
        setup_robot_arm();

        reset();
    };

    virtual ~ApplyForceWorldNode() = default;

    void step_once() {
        customPreStep();
        world_->step();
        customPostStep();
    }

    void reset() {
        Eigen::VectorXd zeros = robot_->getPositions();
        zeros.setZero();
        robot_->setPositions(zeros);
        robot_->setVelocities(zeros);
        robot_->setForces(zeros);

        // "Hip A"
        robot_->setPosition(0, 1.0);
        // "Hip B"
        robot_->setPosition(1, 1.0);
        // "Knee"
        robot_->setPosition(2, -2.0);
[... 166 lines cut ...]
        return false;
    }

private:
    osg::ref_ptr<ApplyForceWorldNode> node_;
};

int main(int argc, char **argv) {
    bool is_ground_plane_fixed = true;

    dart::gui::osg::ImGuiViewer viewer;
    ds::WorldPtr world = std::make_shared<ds::World>();
    osg::ref_ptr<ApplyForceWorldNode> world_node =
        new ApplyForceWorldNode(world, is_ground_plane_fixed);
    viewer.addWorldNode(world_node);
    viewer.addEventHandler(new EventHandler(world_node));
    viewer.setUpViewInWindow(0, 0, 640, 480);
    viewer.getCameraManipulator()->setHomePosition(
        osg::Vec3( 2.57f,  3.14f, 1.64f),
        osg::Vec3( 0.00f,  0.00f, 0.00f),
        osg::Vec3(-0.24f, -0.25f, 0.94f));
    viewer.setCameraManipulator(viewer.getCameraManipulator());
    viewer.run();
    return 0;
}`

azeey, 2023-11-28:
I believe the forces on contacts are computed as part of the `World::step` call by the contact constraint solver. I don't think you can use the collision detector directly and expect to get forces since the collision detector doesn't know anything about forces in the system. Instead, you can use `World::getLastCollisionResult` to get the collision results from the last `World::step` call with forces computed. Here's an example in [`gz-physics`](https://github.com/gazebosim/gz-physics/blob/780e2d0b08a373d586c6a05bf7445094e13daa1c/dartsim/src/SimulationFeatures.cc#L171-L180)

Nickick-ICRS, 2023-11-29:
Ah I see, seems I had missed that. I figured the collision detector would be able to extract the forces from the world, but it seems not.

Either way, this isn't a bug, and you've also provided the correct solution to my problem, so I will mark this as closed.

closed 2023-11-29 by Nickick-ICRS