dartsim/dart #1781: Build fails with gcc-13 due to a warnings

j-rivero, 2024-01-26:
### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: 6.13.1
* OS name and version name(or number):  Debian Sid, Ubuntu Noble
* Compiler name and version number: GCC 13

#### Expected Behavior
Build without warnings 

#### Current Behavior
I found during some updates on Debian Sid and Ubuntu Noble (both uses gcc-13) at least a couple of failures to build due to problems with warnings.

```
[ 13%] Building CXX object dart/CMakeFiles/dart.dir/integration/Integrator.cpp.o
cd /<<PKGBUILDDIR>>/.pybuild/cpython3_3.11/build/dart && /usr/bin/c++ -DDART_ACTIVE_LOG_LEVEL=2 -DDART_HAVE_spdlog=1 -DFMT_HEADER_ONLY=1 -DFMT_SHARED -DSPDLOG_FMT_EXTERNAL -Ddart_EXPORTS -I/<<PKGBUILDDIR>> -I/<<PKGBUILDDIR>>/.pybuild/cpython3_3.11/build -isystem /usr/include/eigen3 -g -O2 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -ffile-prefix-map=/<<PKGBUILDDIR>>=. -flto=auto -ffat-lto-objects -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -fcf-protection -fdebug-prefix-map=/<<PKGBUILDDIR>>=/usr/src/dart-6.13.1+ds-2~osrf2 -Wdate-time -D_FORTIFY_SOURCE=3 -fPIC -Wall -Wextra -fPIC -Werror -MD -MT dart/CMakeFiles/dart.dir/integration/Integrator.cpp.o -MF CMakeFiles/dart.dir/integration/Integrator.cpp.o.d -o CMakeFiles/dart.dir/integration/Integrator.cpp.o -c /<<PKGBUILDDIR>>/dart/integration/Integrator.cpp
In file included from /<<PKGBUILDDIR>>/dart/math/detail/Geometry-impl.hpp:38,
                 from /<<PKGBUILDDIR>>/dart/math/Geometry.hpp:657,
                 from /<<PKGBUILDDIR>>/dart/math/detail/TriMesh-impl.hpp:38,
                 from /<<PKGBUILDDIR>>/dart/math/TriMesh.hpp:119,
                 from /<<PKGBUILDDIR>>/dart/math/TriMesh.cpp:33:
/<<PKGBUILDDIR>>/dart/external/convhull_3d/convhull_3d.h: In function ‘convhull_3d_build’:
/<<PKGBUILDDIR>>/dart/external/convhull_3d/convhull_3d.h:583:33: error: argument 1 range [18446744073709551615, 18446744073709551615] exceeds maximum object size 9223372036854775807 [-Werror=alloc-size-larger-than=]
  583 |   reldist = (CH_FLOAT*)ch_calloc((nVert - d - 1), sizeof(CH_FLOAT));
      |                                 ^
In file included from /usr/include/c++/13/bits/std_abs.h:38,
                 from /usr/include/c++/13/cmath:49,
                 from /usr/include/eigen3/Eigen/src/Core/util/Macros.h:679,
                 from /usr/include/eigen3/Eigen/Core:19,
                 from /<<PKGBUILDDIR>>/dart/math/Mesh.hpp:38,
                 from /<<PKGBUILDDIR>>/dart/math/TriMesh.hpp:38:
/usr/include/stdlib.h:675:14: note: in a call to allocation function ‘calloc’ declared here
  675 | extern void *calloc (size_t __nmemb, size_t __size)
      |              ^~~~~~
[ 13%] Building CXX object dart/CMakeFiles/dart.dir/integration/RK4Integrator.cpp.o
cd /<<PKGBUILDDIR>>/.pybuild/cpython3_3.11/build/dart && /usr/bin/c++ -DDART_ACTIVE_LOG_LEVEL=2 -DDART_HAVE_spdlog=1 -DFMT_HEADER_ONLY=1 -DFMT_SHARED -DSPDLOG_FMT_EXTERNAL -Ddart_EXPORTS -I/<<PKGBUILDDIR>> -I/<<PKGBUILDDIR>>/.pybuild/cpython3_3.11/build -isystem /usr/include/eigen3 -g -O2 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -ffile-prefix-map=/<<PKGBUILDDIR>>=. -flto=auto -ffat-lto-objects -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -fcf-protection -fdebug-prefix-map=/<<PKGBUILDDIR>>=/usr/src/dart-6.13.1+ds-2~osrf2 -Wdate-time -D_FORTIFY_SOURCE=3 -fPIC -Wall -Wextra -fPIC -Werror -MD -MT dart/CMakeFiles/dart.dir/integration/RK4Integrator.cpp.o -MF CMakeFiles/dart.dir/integration/RK4Integrator.cpp.o.d -o CMakeFiles/dart.dir/integration/RK4Integrator.cpp.o -c /<<PKGBUILDDIR>>/dart/integration/RK4Integrator.cpp
cc1plus: all warnings being treated as errors
make[3]: *** [dart/CMakeFiles/dart.dir/build.make:443: dart/CMakeFiles/dart.dir/math/TriMesh.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
In file included from /usr/include/eigen3/Eigen/Core:294,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /<<PKGBUILDDIR>>/dart/math/Geometry.hpp:36,
                 from /<<PKGBUILDDIR>>/dart/math/Geometry.cpp:33:
In member function ‘Eigen::PlainObjectBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >::PlainObjectBase(Eigen::PlainObjectBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >&&)’,
    inlined from ‘Eigen::Matrix<double, 2, 1, 0, 2, 1>::Matrix(Eigen::Matrix<double, 2, 1, 0, 2, 1>&&)’ at /usr/include/eigen3/Eigen/src/Core/Matrix.h:274:30,
    inlined from ‘dart::math::computeClosestPointOnSupportPolygon(unsigned long&, unsigned long&, Eigen::Matrix<double, 2, 1, 0, 2, 1> const&, std::vector<Eigen::Matrix<double, 2, 1, 0, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 2, 1, 0, 2, 1> > > const&)’ at /<<PKGBUILDDIR>>/dart/math/Geometry.cpp:2096:10:
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:496:9: error: ‘MEM[(__m128d * {ref-all})&result]’ may be used uninitialized [-Werror=maybe-uninitialized]
  496 |       : m_storage( std::move(other.m_storage) )
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/<<PKGBUILDDIR>>/dart/math/Geometry.cpp: In function ‘dart::math::computeClosestPointOnSupportPolygon(unsigned long&, unsigned long&, Eigen::Matrix<double, 2, 1, 0, 2, 1> const&, std::vector<Eigen::Matrix<double, 2, 1, 0, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 2, 1, 0, 2, 1> > > const&)’:
/<<PKGBUILDDIR>>/dart/math/Geometry.cpp:2079:25: note: ‘MEM[(__m128d * {ref-all})&result]’ was declared here
 2079 |   Eigen::Vector2d test, result;
      |                         ^~~~~~
cc1plus: all warnings being treated as errors
```

I see a couple of problems:

 * A warning on `convhull_3d.h`:  "exceeds maximum object size". I was looking a little bit into it but seems not trivial to me know whats going on.
 * A warning on `Geometry.cpp` when using Eigen: "may be used uninitialized"

#### Steps to Reproduce
1. Get a Debian-Sid or Ubuntu Noble platforms (docker images maybe) and try to build from DART 6.13.1

#### Code to Reproduce

Probably a dockerfile like should work:

```Dockerfile
FROM ubuntu:noble

RUN echo "deb-src http://archive.ubuntu.com/ubuntu/ noble main universe" > /etc/apt/sources.list.d/ubuntu-deb-src.list \
  && apt-get update \
  && apt-get install -y build-essential git \
  && apt-get build-dep -y dart

RUN git clone https://github.com/dartsim/dart -b v6.13.1 --depth 1 \
  && cd dart \
  && cmake -B build
```

j-rivero, 2024-02-01:
Checking the first error initializing, I think that the compiler complains about the `result` variable in the `Geometry.cpp` file. Sorry I have no idea about what would be a default value for `computeClosestPointOnSupportPolygon` in the case that `_support.size()` is less than 0, just set Zero() below:

```diff
diff --git a/dart/math/Geometry.cpp b/dart/math/Geometry.cpp
index fef4c9c..0b82b95 100644
--- a/dart/math/Geometry.cpp
+++ b/dart/math/Geometry.cpp
@@ -2076,13 +2076,13 @@ Eigen::Vector2d computeClosestPointOnSupportPolygon(
   }
 
   double best = std::numeric_limits<double>::infinity(), check;
-  Eigen::Vector2d test, result;
+  Eigen::Vector2d result = Eigen::Vector2d::Zero();
   for (std::size_t i = 0; i < _support.size(); ++i)
   {
     const Eigen::Vector2d& p1 = (i == 0) ? _support.back() : _support[i - 1];
     const Eigen::Vector2d& p2 = _support[i];
 
-    test = computeClosestPointOnLineSegment(_p, p1, p2);
+    Eigen::Vector2d test = computeClosestPointOnLineSegment(_p, p1, p2);
     check = (test - _p).norm();
     if (check < best)
     {
```

jslee02, 2024-02-01:
The result variable would become N/A when _support.size() is called, so it's acceptable to initialize it to zero. Also, I have found more errors beyond those mentioned in the post, which I am currently addressing in this branch: https://github.com/dartsim/dart/compare/release-6.13...fix_ci

Let me get back to you once the changes are ready. Sorry for taking too long!

j-rivero, 2024-02-01:
> I'll make sure to get back to you once the changes are ready. Sorry for taking too long!

No worries, sorry for not being able to provide more help this time.

jslee02, 2024-03-16:
I have finally resolved the build issues with gcc-13 by testing it on Ubuntu 24.04 LTS in version 6.14 (e.g., [build log](https://github.com/dartsim/dart/actions/runs/8289208167/job/22685214654)). I'd prefer to release this as version 6.14, but I'm open to creating a patch for version 6.13 if needed. Just let me know.

jslee02, 2024-03-16:
No new features to 6.14, but the supported platforms in 6.14 are:

- Ubuntu Focal on amd64 / GCC 9.4 / amd64
- Ubuntu Jammy on amd64 / GCC 11.3 / amd64
- macOS 12 (Monterey) / Clang 14 / amd64
- Windows / MSVC 19.37 / amd64

and breaking changes (so far):

- Removed planning component

jslee02, 2024-03-17:
Never mind, I've created a patch for 6.13 (#1793)! Let me create a patch release with this.

PR #1793 (2024-03-17): Fix build errors with gcc >= 13
Resolves #1781

***

#### Before creating a pull request

- [ ] Document new methods and classes
- [ ] Format new code files using ClangFormat by running `make format`
- [ ] Build with `-DDART_TREAT_WARNINGS_AS_ERRORS=ON` and resolve all the compile warnings

#### Before merging a pull request

- [ ] Set version target by selecting a milestone on the right side
- [ ] Summarize this change in `CHANGELOG.md`
- [ ] Add unit test(s) for this change
- [ ] Add Python bindings for new methods and classes
closed 2024-03-17 by jslee02