dartsim/dart #1794: Main Documentation Website Down

Nickick-ICRS, 2024-03-19:
### Bug Report
I am unable to open the class / API documentation website (https://dartsim.github.io/dart/). I also am unable to access it from e.g. a link from the biped tutorial (https://dartsim.github.io/tutorials_biped.html).

Obviously this is problematic.

#### Environment
* DART version: [any]
* OS name and version name(or number): Ubuntu 20.04, Mozilla Firefox
* Compiler name and version number: N/A

#### Expected Behavior
Documentation opens and I can look at the API.

#### Current Behavior
GitHub pages 404 page.

#### Steps to Reproduce
Go to https://dartsim.github.io/dart/

#### Code to Reproduce
Not applicable

matthias-mayr, 2024-03-19:
Not a fix, but some sort of immediate workaround for people who just want to read the page right now:  
The internet archive had saved a version last May: [Link](https://web.archive.org/web/20230519192656/https://dartsim.github.io/tutorials_biped.html)

jslee02, 2024-03-20:
The issue should be resolved at this point: https://dartsim.github.io/dart/ Thank you for bringing this to our attention!

closed 2024-03-20 by jslee02