dartsim/dart #1795: ABI of the library depends on the value of NDEBUG?

traversaro, 2024-03-23:
It seems that currently the ABI of the library depends on the value of `NDEBUG` used when using the headers of the library, see:
* https://github.com/dartsim/dart/blob/2437749590b778074434d3a0f6208bf366ac12af/dart/constraint/PgsBoxedLcpSolver.hpp#L83
* https://github.com/dartsim/dart/blob/2437749590b778074434d3a0f6208bf366ac12af/dart/constraint/PGSLCPSolver.hpp#L61
* https://github.com/dartsim/dart/blob/2437749590b778074434d3a0f6208bf366ac12af/dart/common/MemoryManager.hpp#L154

However, this value could be different in the downstream project using dart w.r.t. to the value used when compiling dart, for example see https://github.com/conda-forge/gz-physics-feedstock/pull/26#issuecomment-2016526660 . 

Probably it could make sense to save in https://github.com/dartsim/dart/blob/2437749590b778074434d3a0f6208bf366ac12af/dart/config.hpp.in#L65 the `NDEBUG` value used when compiling dart (or something similar) and use that in the `.hpp` ?

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: 6.13.2
* OS name and version name(or number): Not relevant
* Compiler name and version number: Not relevant

#### Expected Behavior
> Please describe the behavior you are expecting.

#### Current Behavior

> What is the current behavior?

See description.

#### Steps to Reproduce
Not relevant?

#### Code to Reproduce
> Please remember that with a working code it's easier to reproduce the bug and it's much faster to fix it.
> It would be great to fork this repo and create a regression test using [this template](https://github.com/dartsim/dart/blob/main/tests/regression/test_Issue000Template.cpp).

Not relevant?

jslee02, 2024-03-24:
Good suggestion! Created https://github.com/dartsim/dart/pull/1796

jslee02, 2024-03-30:
I believe #1796 has resolved this issue. If there are more things that need to be resolved, please reopen it or create a new one. Thank you!

PR #1796 (2024-03-24): Replace NDEBUG with custom flags to avoid ABI issues
- Replaced NDEBUG with custom flags to avoid ABI issues for https://github.com/dartsim/dart/issues/1795.
- Added DART_ prefix to avoid potential name conflict.
- Consolidated custom flags for release mode into DART_BUILD_MODE_RELEASE to simplify, based on usage.
- Included `config.hpp` whenever needed
- Fixed typos

Resolves #1795

***

#### Before creating a pull request

- [x] Document new methods and classes
- [ ] Format new code files using ClangFormat by running `make format`
- [x] Build with `-DDART_TREAT_WARNINGS_AS_ERRORS=ON` and res [... cut]
closed 2024-03-30 by jslee02