dartsim/dart #1797: Some tests fail

yurivict, 2024-04-04:
### Bug Report
```
Test project /usr/ports/misc/dartsim/work/.build
      Start  1: INTEGRATION_Aspect
 1/72 Test  #1: INTEGRATION_Aspect ................   Passed    0.03 sec
      Start  2: INTEGRATION_Building
 2/72 Test  #2: INTEGRATION_Building ..............   Passed    0.03 sec
      Start  3: INTEGRATION_Common
 3/72 Test  #3: INTEGRATION_Common ................   Passed    2.17 sec
      Start  4: INTEGRATION_Concurrency
 4/72 Test  #4: INTEGRATION_Concurrency ...........   Passed    0.03 sec
      Start  5: INTEGRATION_Constraint
 5/72 Test  #5: INTEGRATION_Constraint ............   Passed    0.04 sec
      Start  6: INTEGRATION_Frames
 6/72 Test  #6: INTEGRATION_Frames ................   Passed    0.04 sec
      Start  7: INTEGRATION_Friction
 7/72 Test  #7: INTEGRATION_Friction ..............   Passed    0.13 sec
      Start  8: INTEGRATION_InverseKinematics
 8/72 Test  #8: INTEGRATION_InverseKinematics .....   Passed    0.03 sec
      Start  9: INTEGRATION_NameManagement
 9/72 Test  #9: INTEGRATION_NameManagement ........   Passed    0.03 sec
      Start 10: INTEGRATION_Signal
10/72 Test #10: INTEGRATION_Signal ................   Passed    0.03 sec
      Start 11: test_CollisionGroups
11/72 Test #11: test_CollisionGroups ..............***Failed    0.06 sec
      Start 12: test_ConstraintSolver
12/72 Test #12: test_ConstraintSolver .............   Passed    0.03 sec
      Start 13: test_ContactConstraint
13/72 Test #13: test_ContactConstraint ............   Passed    0.06 sec
      Start 14: test_GenericJoints
14/72 Test #14: test_GenericJoints ................   Passed    0.03 sec
      Start 15: test_Inertia
15/72 Test #15: test_Inertia ......................   Passed    0.03 sec
      Start 16: test_Lemke
16/72 Test #16: test_Lemke ........................   Passed    0.03 sec
      Start 17: test_LocalResourceRetriever
17/72 Test #17: test_LocalResourceRetriever .......   Passed    0.03 sec
      Start 18: test_Optimizer
18/72 Test #18: test_Optimizer ....................   Passed    0.08 sec
      Start 19: test_ScrewJoint
19/72 Test #19: test_ScrewJoint ...................   Passed    0.04 sec
      Start 20: test_Subscriptions
20/72 Test #20: test_Subscriptions ................   Passed    0.03 sec
      Start 21: test_CompositeResourceRetriever
21/72 Test #21: test_CompositeResourceRetriever ...   Passed    0.04 sec
      Start 22: test_DartResourceRetriever
22/72 Test #22: test_DartResourceRetriever ........   Passed    0.04 sec
      Start 23: test_FileInfoWorld
23/72 Test #23: test_FileInfoWorld ................   Passed    0.08 sec
      Start 24: test_PackageResourceRetriever
24/72 Test #24: test_PackageResourceRetriever .....   Passed    0.04 sec
      Start 25: test_MjcfParser
25/72 Test #25: test_MjcfParser ...................   Passed    0.33 sec
      Start 26: test_SdfParser
26/72 Test #26: test_SdfParser ....................   Passed    0.50 sec
      Start 27: test_SkelParser
27/72 Test #27: test_SkelParser ...................   Passed    0.23 sec
      Start 28: test_VskParser
28/72 Test #28: test_VskParser ....................   Passed    0.10 sec
      Start 29: test_DartLoader
29/72 Test #29: test_DartLoader ...................***Failed    0.05 sec
      Start 30: test_IkFast
30/72 Test #30: test_IkFast .......................***Failed    0.05 sec
      Start 31: test_NearestNeighbor
31/72 Test #31: test_NearestNeighbor ..............   Passed    0.04 sec
      Start 32: test_Distance
32/72 Test #32: test_Distance .....................   Passed    0.05 sec
      Start 33: test_Raycast
33/72 Test #33: test_Raycast ......................***Exception: SegFault  0.05 sec
      Start 34: test_ForceDependentSlip
[... 274 lines cut ...]
  Actual: false
Expected: true
[  FAILED  ] DartLoader.parseVisualCollisionName (0 ms)
[ RUN      ] DartLoader.Options
[       OK ] DartLoader.Options (1 ms)
[----------] 17 tests from DartLoader (6 ms total)

[----------] Global test environment tear-down
[==========] 17 tests from 1 test case ran. (6 ms total)
[  PASSED  ] 11 tests.
[  FAILED  ] 6 tests, listed below:
[  FAILED  ] DartLoader.parseSkeleton_LoadsPrimitiveGeometry
[  FAILED  ] DartLoader.parseJointProperties
[  FAILED  ] DartLoader.mimicJoint
[  FAILED  ] DartLoader.badMimicJoint
[  FAILED  ] DartLoader.KR5MeshColor
[  FAILED  ] DartLoader.parseVisualCollisionName

 6 FAILED TESTS
```

#### Environment
* DART version: 6.13.2
* OS name and version name(or number): FreeBSD 14.0
* Compiler name and version number: clang-17

jslee02, 2024-04-04:
Thank you for reporting the test failures. While FreeBSD isn't directly supported, I'm interested in ensuring compatibility. Unfortunately, I don't have access to a FreeBSD environment. If you could provide a Dockerfile for FreeBSD to reproduce this issue, it would be conducive to reproducing and debugging it!

yurivict, 2024-04-04:
Unfortunately, Docker doesn't support FreeBSD.

I have these instructions on how to quickly get a running FreeBSD environment:

> Is there an easy way to test under FreeBSD?

1. Install the [FreeBSD VM image](https://download.freebsd.org/ftp/releases/VM-IMAGES/14.0-RELEASE/amd64/Latest/) into VirtualBox.
2. Boot FreeBSD
3. Install git: pkg install git
4. Check out the ports tree: git clone https://git.FreeBSD.org/ports.git /usr/ports
5. cd /usr/ports/misc/dartsim
6. Install dependencies: pkg install -A ``` `make missing` ```
7. Build: make 
8. Run tests: make test

jslee02, 2024-04-05:
Thanks for the instructions. I haven't used VirtualBox or FreeBSD before, so setting this up may take some time, and I can't promise a specific timeline, but contributions from others are welcome in the meantime!

yurivict, 2024-06-26:
6.14.0 has these tests failing:

```
The following tests FAILED:
         11 - test_CollisionGroups (Failed)
         29 - test_DartLoader (Failed)
         30 - test_IkFast (Failed)
         33 - test_Raycast (ILLEGAL)
         35 - test_Collision (Failed)
         43 - test_ForwardKinematics (SEGFAULT)
         48 - test_Issue838 (SEGFAULT)
         54 - test_Issue1184 (Failed)
```

[log](https://freebsd.org/~yuri/dartsim-6.14.0-tests-failures-LastTest.log)

jslee02, 2024-06-26:
Thank you for reporting this issue. Unfortunately, I currently don't have the bandwidth to debug this on FreeBSD. Most of the failures seem to be related to dependencies, which might be due to how they are packaged on FreeBSD. As a possible workaround, some tests might succeed if you disable building with Bullet.