dartsim/dart #1804: 6.13.2: Build fails: "osg.hpp: No such file or directory"

heeplr, 2024-05-10:
### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [ ] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
  - that discourse URL gives unknown host error
- [X] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: 6.13.2 source tarball
* OS name and version name(or number): gentoo linux
* Compiler name and version number: gcc 13.2.1

#### Expected Behavior
Successful build.

#### Current Behavior
```
/var/tmp/portage_notmpfs/portage/sci-electronics/dart-6.13.2/work/dart-6.13.2/python/dartpy/gui/osg/DragAndDrop.cpp                                                                                                                            
/var/tmp/portage_notmpfs/portage/sci-electronics/dart-6.13.2/work/dart-6.13.2/python/dartpy/gui/osg/DragAndDrop.cpp:34:10: fatal error: dart/gui/osg/osg.hpp: No such file or directory                                                                                                                                                                           
   34 | #include <dart/gui/osg/osg.hpp>                                                                                                                                          
      |          ^~~~~~~~~~~~~~~~~~~~~~                                                                                                                                          
compilation terminated
```

The file is not generated from the osg.hpp.in and I'm not sure why.

Full build log: [build.log](https://github.com/dartsim/dart/files/15275182/build.log)

#### Steps to Reproduce
1. unpack source tarball
2. compile as described in [docs](http://dartsim.github.io/install_dart_on_archlinux.html#build-and-install-dart)

heeplr, 2024-05-10:
I found that this doesn't happen when OpenSceneGraph is installed.

Leaving this open since it's probably some cmake issue not properly respecting the unfound optional osg package.

jslee02, 2024-06-27:
Thanks for reporting! This should be fixed in 6.14.0 since now dartpy isn't built when the required components are not available (e.g., OSG is not detected):

https://github.com/dartsim/dart/blob/d2f56b4abf347afbbc92cff763ec8a9dcf7c1fb1/python/dartpy/CMakeLists.txt#L45-L56

closed 2024-06-27 by jslee02