dartsim/dart #1811: trouble in build v6.13.0 fails on Ubuntu 20.04

abc8512, 2024-06-07:
> If you're not reporting a bug, please use the [forum](https://dartsim.discourse.group/) to ask questions.

Is there anyone who can help me build?
It seems that forum is not reachable.
I tried both 6.13.1 and 6.13.0 but I have the same issue.

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [ ] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: v6.13.0
* OS name and version name(or number): Ubunut 20.04
* Compiler name and version number: GCC 9.4.0

#### Expected Behavior
> Please describe the behavior you are expecting.

#### Current Behavior
> What is the current behavior?
> If you have trouble in `cmake` or `make`, it would be very helpful to build DART with verbose option:
> `$ cmake .. -DDART_VERBOSE=On`
> `$ make VERBOSE=1`
> and then dump the results into a [gist](https://gist.github.com/) and share the link to gist here.

When I run make, it complains about c++17. I have GCC 9.4.0. I think it should work.
https://gist.github.com/abc8512/1c7ac07537a2ca6d5561fc59b14a0c71.js
I fount -std=c++14 at line 682

#### Steps to Reproduce
> Please provide detailed steps for reproducing the issue.
1. git clone
2. git checkout tags/v6.13.0
3. cmake .. -DCMAKE_INSTALL_PREFIX=/usr/ -DCMAKE_BUILD_TYPE=Release -DDART_BUILD_DARTPY=ON -DDART_VERBOSE=On
4. make VERBOSE=1

#### Code to Reproduce
> Please remember that with a working code it's easier to reproduce the bug and it's much faster to fix it.
> It would be great to fork this repo and create a regression test using [this template](https://github.com/dartsim/dart/blob/main/tests/regression/test_Issue000Template.cpp).

jslee02, 2024-06-27:
Hi @abc8512, due to limited bandwidth, I'm currently unable to test every dev environment. Do you need to build a particular version on a specific Ubuntu version? If not, could you please build the latest release version (6.14.1) on a tested Ubuntu environment (22.04 or 24.04) or even use Pixi?

jslee02, 2024-07-05:
I am closing this as no further action is planned. Feel free to reopen if you have any other suggestions or ideas for moving forward.

abc8512, 2024-07-07:
Hi @jslee02 , I have no plan to upgrade my Ubuntu version yet. I will suggest that my team moves to 22.04 or 24.04. Thanks!