dartsim/dart #1839: MeshShape::setScale does not accept negative values?

costashatz, 2024-08-29:
I do not understand why `MeshShape::setScale()` [does not accept negative values](https://github.com/dartsim/dart/blob/main/dart/dynamics/MeshShape.cpp#L239). This is a common practice to mirror the mesh (otherwise we need to have multiple "duplicate" meshes, especially when using the URDF format).

Also, this check was "recently" added (I did not check when this happened, but it wasn't there for many years that I am using the library). Is there a specific rationale/guideline for this? Imho, we should allow negative scaling.

jslee02, 2024-08-29:
Accepting negative values makes sense to me. We may need to ensure that it works fine with the codebase (e.g., [updating bounding box](https://github.com/dartsim/dart/blob/cac87c136070a374b4705983c5a3a29db544d0e2/dart/dynamics/MeshShape.cpp#L358-L361)) and Assimp also allows it.

Regarding when and why it was changed, I don't recall quite frankly. The git log says that line was changed 6 years ago (from checking individual xyz components to using Eigen's all() function) and the previous commit of the 6 years ago also didn't accept negative values.

[image]

[image]

costashatz, 2024-08-29:
Oh I see what happened. I was compiling with disabled assertions!

In any case, we should allow for negative scaling. Apart from the `bounding box update`, where else should we handle this?

jslee02, 2024-08-31:
It seems only the updating bounding box. #1841 to continue this discussion!

jslee02, 2024-09-03:
Closing per #1841. Please let me know if this works for you!

PR #1841 (2024-09-03): Allow negative scale for MeshShape
This PR supports negative scaling in MeshShape as requested #1839. While it's not extensively tested yet, initial results are promising. Scaling the pelvis mesh by positive (5) and negative (-5) values worked as expected in preliminary tests.

Pelvis with scale(5, 5, 5):

https://github.com/user-attachments/assets/9bb243a1-ba4b-479c-944d-cb2de7541b31

Pelvis with scale(-5, -5, -5):

https://github.com/user-attachments/assets/81ada3b6-6f55-40bd-bd35-0dbeb49fd3a8

Resolves #1839.

***

#### Before creating a pull request

- [ ] Document new methods and classes
 [... cut]
closed 2024-09-03 by jslee02