dartsim/dart #1898: imgui inputText/float/int duplicated characters

XDinEuro, 2025-01-10:
Taking a build in example, `point_cloud`
[image]

If I focus on the text input, and type 1, it will appear 11 in input box, same as all characters (except backspace).

I checked the origin [imgui](https://github.com/ocornut/imgui) project, it looks fine in its example.

I wonder if we used it in a wrong way ?

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: 6.14.4
* OS name and version name(or number): ubuntu22.04
* Compiler name and version number: gcc 11.4

#### Expected Behavior
The input box shows exactly what I typed.

#### Current Behavior
The input characters are duplicated.

#### Steps to Reproduce
> Please provide detailed steps for reproducing the issue.
1. build the project and examples 
2. open point_cloud
3. focus on the input box by single click on it
4. type a number, like '1'

#### Code to Reproduce

jslee02, 2025-01-14:
Could you retry the build with `-DDART_USE_SYSTEM_IMGUI=ON` in CMake? The vendored imgui is a bit outdated.

XDinEuro, 2025-01-15:
hey, I managed it by dong this: replace the imgui source files with up to date files from [https://github.com/ocornut/imgui/](https://github.com/ocornut/imgui/), and keep the changes marked with MODIFIED FOR DART in [https://github.com/dartsim/dart/blob/main/dart/external/imgui/imgui_impl_opengl2.cpp#L120](https://github.com/dartsim/dart/blob/main/dart/external/imgui/imgui_impl_opengl2.cpp#L120) and resolve some compile errors. Eventually I made it with some effort.

Is that make sense to update the imgui in dart ?

jslee02, 2025-01-20:
I'm glad to hear you found a workaround! Eventually, I plan to remove the IMGUI source from this repository. Originally, it was included because (1) IMGUI wasn’t available through package managers, and (2) modifications marked as "MODIFIED_FOR_DART" were necessary. However, these reasons no longer seem to apply.

To support this transition, I’ve added the `DART_USE_SYSTEM_IMGUI` option, allowing for an external IMGUI installation as a long-term solution.

Updating the IMGUI source in this repository is fine. Feel free to submit a PR for the update, but I would recommend using the `DART_USE_SYSTEM_IMGUI` option instead.

XDinEuro, 2025-02-06:
In my case MODIFIED_FOR_DART is necessary, otherwise the `Interactive frame` can't be visualized properly. You could try the demo `fetch`. 

But it make much more sense to use DART_USE_SYSTEM_IMGUI option. I will close the issue. Thanks for the support !

closed 2025-02-06 by XDinEuro