dartsim/dart #1915: test_ForceDependentSlip is failing in Debian Sid after updating ode from 0.16.2 to 0.16.6

j-rivero, 2025-02-06:
While working in the Debian transition to DART 0.16.2 the `test_ForceDependentSlip` is failing in Debian Sid after the distribution updated ode from 0.16.2 to 0.16.6.

Note that the test expectations were already relaxed for the armel architecture https://salsa.debian.org/science-team/dart/-/blob/master/debian/patches/fix-test_ForceDependentSlip-armel.patch?ref_type=heads

Notes:
  *  [0.16.6 to 0.16.2 commits diff in upstream ode](https://bitbucket.org/odedevs/ode/branches/compare/0.16.6%0D0.16.2#commits)
  * [0.16.6 to 0.16.2 changelog diff in upstream ode](https://bitbucket.org/odedevs/ode/branches/compare/0.16.6%0D0.16.2#chg-CHANGELOG.txt)

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [x] I checked the [documentation](http://dartsim.github.io/) and [Discussions](https://github.com/dartsim/dart/discussions) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version:  6.13.2
* OS name and version name(or number): Debian unstable
* Compiler name and version number: gcc-14.2.0

#### Expected Behavior
Test to pass. It was passing using the system ode version 0.16.2

#### Current Behavior
Test is failing: 
```
Running main() from /usr/src/googletest/googletest/src/gtest_main.cc
[==========] Running 2 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 2 tests from ForceDependentSlip
[ RUN      ] ForceDependentSlip.BoxSlipVelocity
[       OK ] ForceDependentSlip.BoxSlipVelocity (113 ms)
[ RUN      ] ForceDependentSlip.CylinderSlipVelocity
./unittests/integration/test_ForceDependentSlip.cpp:265: Failure
The difference between extForceX * slip and body1->getLinearVelocity().x() is 0.00043832661683589763, which exceeds 1e-4, where
extForceX * slip evaluates to 0.02,
body1->getLinearVelocity().x() evaluates to 0.020438326616835898, and
1e-4 evaluates to 0.0001.

./unittests/integration/test_ForceDependentSlip.cpp:266: Failure
The difference between 0.0 and body1->getLinearVelocity().y() is 0.005297929413959148, which exceeds 1e-4, where
0.0 evaluates to 0,
body1->getLinearVelocity().y() evaluates to 0.005297929413959148, and
1e-4 evaluates to 0.0001.

./unittests/integration/test_ForceDependentSlip.cpp:265: Failure
The difference between extForceX * slip and body1->getLinearVelocity().x() is 0.00035668589428858657, which exceeds 1e-4, where
extForceX * slip evaluates to 0.02,
body1->getLinearVelocity().x() evaluates to 0.020356685894288587, and
1e-4 evaluates to 0.0001.

./unittests/integration/test_ForceDependentSlip.cpp:266: Failure
The difference between 0.0 and body1->getLinearVelocity().y() is 0.0043113150438673625, which exceeds 1e-4, where
0.0 evaluates to 0,
body1->getLinearVelocity().y() evaluates to 0.0043113150438673625, and
1e-4 evaluates to 0.0001.

./unittests/integration/test_ForceDependentSlip.cpp:265: Failure
The difference between extForceX * slip and body1->getLinearVelocity().x() is 0.00024724163885008962, which exceeds 1e-4, where
extForceX * slip evaluates to 0.02,
body1->getLinearVelocity().x() evaluates to 0.02024724163885009, and
1e-4 evaluates to 0.0001.

./unittests/integration/test_ForceDependentSlip.cpp:266: Failure
The difference between 0.0 and body1->getLinearVelocity().y() is 0.002784063048611347, which exceeds 1e-4, where
0.0 evaluates to 0,
body1->getLinearVelocity().y() evaluates to 0.002784063048611347, and
1e-4 evaluates to 0.0001.

./unittests/integration/test_ForceDependentSlip.cpp:265: Failure
The difference between extForceX * slip and body1->getLinearVelocity().x() is 0.00011317467529363065, which exceeds 1e-4, where
extForceX * slip evaluates to 0.02,
body1->getLinearVelocity().x() evaluates to 0.020113174675293631, and
1e-4 evaluates to 0.0001.

./unittests/integration/test_ForceDependentSlip.cpp:266: Failure
The difference between 0.0 and body1->getLinearVelocity().y() is 0.00050173758912424411, which exceeds 1e-4, where
0.0 evaluates to 0,
body1->getLinearVelocity().y() evaluates to 0.00050173758912424411, and
1e-4 evaluates to 0.0001.
```

#### Steps to Reproduce
> Please provide detailed steps for reproducing the issue.

A dockerfile.

 ```dockerfile
FROM debian:sid
ENV LANG C
ENV LC_ALL C
ARG DEBIAN_FRONTEND=noninteractive
RUN apt-get update \
    && apt-get install -y \
      build-essential \
      git-buildpackage \
      libdart-dev \
      architecture-is-64-bit \
      cmake \
      coinor-libipopt-dev \
      doxygen \
      dh-sequence-python3 \
      libode-dev \
      libnlopt-cxx-dev \
      libxi-dev \
      libxmu-dev \
      libtinyxml-dev \
      libtinyxml2-dev \
      liburdfdom-dev \
      libopenthreads-dev \
      libopenscenegraph-dev \
      libspdlog-dev \
      libfmt-dev \
      freeglut3-dev \
      googletest \
      pybind11-dev \
      python3-all-dev \
      python3-numpy \
      python3-pytest \
      jdupes \
   && rm -rf /var/lib/apt/lists/*

RUN git clone https://salsa.debian.org/science-team/dart \
    && cd dart \
    && gbp buildpackage -j8
```

#### Code to Reproduce
> Please remember that with a working code it's easier to reproduce the bug and it's much faster to fix it.
> It would be great to fork this repo and create a regression test using [this template](https://github.com/dartsim/dart/blob/main/tests/regression/test_Issue000Template.cpp).

traversaro, 2025-02-06:
I think it is due to a change in ode upstream, see https://bitbucket.org/odedevs/ode/issues/87/box-cylinder-ccd-check-disabled-by-default . See https://github.com/conda-forge/dartsim-feedstock/issues/62 for a bit more context.

closed 2025-02-06 by j-rivero

lepalom, 2025-02-06:
Sorry, but I think that I'm missing something because I don't understand the issue. I have rebuild ode 0.16.6

traversaro, 2025-02-06:
> Sorry, but I think that I'm missing something because I don't understand the issue. I have rebuild ode 0.16.6

I am not sure why, but the `test_ForceDependentSlip` dartsim test never passed when using ODE configured to use libccd for the box cylinder collision. 

If I recall correctly, Debian uses the autotools build system to build libode, and in libode 0.16.2 the autotools defaulted to not use libccd for the box cylinder collision, even if this was a inconsistency w.r.t. to the CMake build system of ODE, that is the issue reported https://bitbucket.org/odedevs/ode/issues/87/box-cylinder-ccd-check-disabled-by-default. The inconsistency was solved by the upstream ODE mantainers in https://bitbucket.org/odedevs/ode/commits/11fad5231ac7cab42dec7b34c1f121376216098e by using libccd for box cylinder collision also when using the autotools build system, see the changelog description:

          Autotools build scripts, when configured with "--enable-libccd",
          will now assign all the relevant colliders to the libccd. 
          Before, it used to keep default collider for Box-Cylinder pair and 
          that was inconsistent with other configuration methods (CMake and 
          premake4.exe). To obtain the old behavior configure as 
          "--enable-libccd --with-box-cylinder=default".

This change was released in libode 0.16.5 . 

Interestingly, before the upstream change to align autotools with cmake, we did the reverse change in the conda-forge packaging of libode in https://github.com/conda-forge/libode-feedstock/pull/23, and this is the reason why the test pass in dartsim+conda-forge CI.

j-rivero, 2025-02-06:
Silvio arrived 6 minutes before me with the same changelog entry info that I found relevant.

> If I recall correctly, Debian uses the autotools build system to build libode

This is correct, I was checking the logs https://salsa.debian.org/games-team/ode/-/jobs/6997942 (latest 0.16.6).

In summary @lepalom , I think that we need to add `--with-box-cylinder=default` to the d/rules file. Thanks guys for the help.

lepalom, 2025-02-06:
Yes, @j-rivero and @traversaro . I built again with that option and I worked. First I did it with --with-box-cylinder=libccd because it was what I misunderstood and obviously it failed.

I push the new version of ode soon.

lepalom, 2025-02-07:
ode_0.16.6-2_source.changes ACCEPTED into unstable 

@j-rivero I think that you can test dark and push it to unstable.

j-rivero, 2025-02-10:
> [@j-rivero](https://github.com/j-rivero) I think that you can test dark and push it to unstable.

Thanks Leo! It worked, I've uploaded Dart to unstable without the test failure. Thanks everyone.