dartsim/dart #1928: Repository size is very large, possibly due to gh-pages branch

scpeters, 2025-03-07:
Recently it takes a very long time to clone this repository. I suspect it may be due to the `gh-pages` branch. I was able to clone `release-6.15` with `--single-branch` fairly quickly, but it takes a long time to clone `gh-pages` with `--single-branch`.

jslee02, 2025-03-07:
Yeah, that might be the reason for the slow cloning. I've just cleaned up the history of the gh-pages branch using the following commands:

```
git checkout gh-pages
git checkout --orphan gh-pages-new
git add -A
git commit -m "Fresh start: Preserved latest gh-pages content"
git push --force origin gh-pages-new:gh-pages
```

After this, I was able to clone with the full history in about 9 seconds:

```
➜  dart_clone time git clone https://github.com/dartsim/dart
Cloning into 'dart'...
remote: Enumerating objects: 119596, done.
remote: Counting objects: 100% (27746/27746), done.
remote: Compressing objects: 100% (5220/5220), done.
remote: Total 119596 (delta 22296), reused 26889 (delta 22176), pack-reused 91850 (from 3)
Receiving objects: 100% (119596/119596), 120.32 MiB | 16.06 MiB/s, done.
Resolving deltas: 100% (92489/92489), done.
git clone https://github.com/dartsim/dart  8.54s user 0.69s system 86% cpu 10.709 total
```

scpeters, 2025-03-10:
thanks! it's much smaller now to clone everything

closed 2025-03-10 by scpeters